mcknly/breadboard-os #1: User on RaspberryPi forums unable to build BreadBoard-OS

DeeJay, 2024-05-16:
May I draw your attention to a recent posting on the RaspberryPi forums from a user who is unable to build BreadBoard-OS from this repo.

[https://forums.raspberrypi.com/viewtopic.php?t=370922](https://forums.raspberrypi.com/viewtopic.php?t=370922)

I am not the Originating Poster in that thread. I have not attempted to replicate the issue reported there. I think it is safe to say from my previous interactions with them in that forum that they have a substantial amount of experience with using the RPi Pico SDK so their report is probably accurate and reliable.

mcknly, 2024-05-16:
@DeeJay I appreciate you raising this to my attention! We will troubleshoot on that forum post and then add the resolution here.

DeeJay, 2024-05-16:
With apologies, it seems that thread has now progressed. I came back with the intention of Deleting/Closing the Issue.

Thanks for your prompt acknowledgement and interest.

closed 2024-05-16 by DeeJay

mcknly, 2024-05-16:
@DeeJay the help was much appreciated.