mcknly/breadboard-os #10: Custom driver code not being loaded

devramsean0, 2024-06-05:
Hi!
I'm trying to use breadboard-OS to display some numbers on a 4 digit 8 segment display that uses a 74HC595.

I've added it to CMakeList, the device_drives.c file and the device_drives.h file in the same manner that both other examples are added.
However, it never appears under lib/ like the others. Any ideas?

My code is below:
```c
#include "pico/stdlib.h"
#include "hardware/spi.h"
#include "device_drivers.h"
#include "hardware_config.h"

#define SPI_PORT spi0
#define PIN_CS 17
#define PIN_SCK 18
#define PIN_MOSI 19

#define CMD_DISPLAY_CONTROL 0x88 // Example command, consult datasheet for exact values
#define CMD_DISPLAY_ADDRESS 0xC0 // Example command to set address

// Initialize the SPI interface and the 8-segment LED display
int pico_8seg_led_init(void) {
    // Initialize SPI port
/*     spi_init(SPI_PORT, 500 * 1000); // 500 kHz
    gpio_set_function(PIN_SCK, GPIO_FUNC_SPI);
    gpio_set_function(PIN_MOSI, GPIO_FUNC_SPI);
    
    // Initialize CS pin
    gpio_init(PIN_CS);
    gpio_set_dir(PIN_CS, GPIO_OUT);
    gpio_put(PIN_CS, 1); */
    return 1;
}

// Send a command to the 8-segment LED display
void pico_8seg_led_send_command(uint8_t command) {
    gpio_put(PIN_CS, 0);
    spi_write_blocking(SPI_PORT, &command, 1);
    gpio_put(PIN_CS, 1);
}

// Send data to the 8-segment LED display
void pico_8seg_led_send_data(uint8_t address, uint8_t data) {
    uint8_t buf[2] = {address, data};
    gpio_put(PIN_CS, 0);
    spi_write_blocking(SPI_PORT, buf, 2);
    gpio_put(PIN_CS, 1);
}
```

devramsean0, 2024-06-05:
That fixed it! Thanks :)

closed 2024-06-05 by devramsean0

mcknly, 2024-06-06:
@devramsean0 , FYI, BBOS has hardware-abstracted interfaces for the RP2040 hw including SPI, as long as `HW_USE_SPI0` is set to `true` in `hardware_config.h` (it is by default), it will initialize the pins at boot, and all you have to do is use the `spi0_write_register()` and `spi0_read_registers()` functions without worrying about any pin config stuff. Thanks for checking out the project!

devramsean0, 2024-06-06:
Unfortunaly, it's using spi1 lol. That version of the driver code was wrong