mcknly/breadboard-os #12: Pico_w LED remains off

mcknly, 2024-06-06:
### Discussed in https://github.com/mcknly/breadboard-os/discussions/11

<div type='discussions-op-text'>

<sup>Originally posted by **fatdollar** June  6, 2024</sup>
Yesterday I pulled down all the code and was watching your tutorial video trying to follow along.  I have an issue though were even when trying to turn on the LED it will always say LED is OFF.  Did I build the project incorrectly?

[image]
I've attached a screen shot of the terminal and here is the `project.cmake` file:

```
 # PROJECT NAME - in quotes, no spaces
 set(PROJ_NAME "bbos-dev")
 
 # PROJECT VERSION - in quotes, no spaces, can contain alphanumeric if necessary
 set(PROJ_VER "0.1")
 
 # CLI INTERFACE - 0: use UART for CLI (default), 1: use USB for CLI
 set(CLI_IFACE 1)
 
 # MCU PLATFORM - set the MCU platform being used (i.e. the subdir in 'hardware/')
 set(PLATFORM rp2040)
 
 # BOARD - set the board being used (i.e. 'pico', 'pico_w', etc)
 set(BOARD pico_w)
```

I only worry about this because I wonder if the project is setup correctly to work with the other hardware on the board as well.

Any help would be appreciated. Thanks.
(Excellent project btw)</div>

fatdollar, 2024-06-07:
by way of update I have found the issue with the LED but still working on a fix.  I figured I'd let you know since you're more familiar with the project but the `-DUSING_PICOW` is not being used when building the `cli` directory so during build we get the following:
```
[ 27%] Building C object cli/CMakeFiles/cli.dir/node_dev.c.obj
cd /home/fatdollar/breadboard-os/build/cli && /usr/bin/arm-none-eabi-gcc -DBOARD=pico_w -DCLI_USE_USB=1 -DFREE_RTOS_KERNEL_SMP=1 -DLIB_FREERTOS_KERNEL=1 -DLIB_PICO_BIT_OPS=1 -DLIB_PICO_BIT_OPS_PICO=1 -DLIB_PICO_DIVIDER=1 -DLIB_PICO_DIVIDER_HARDWARE=1 -DLIB_PICO_DOUBLE=1 -DLIB_PICO_DOUBLE_PICO=1 -DLIB_PICO_FLOAT=1 -DLIB_PICO_FLOAT_PICO=1 -DLIB_PICO_INT64_OPS=1 -DLIB_PICO_INT64_OPS_PICO=1 -DLIB_PICO_MALLOC=1 -DLIB_PICO_MEM_OPS=1 -DLIB_PICO_MEM_OPS_PICO=1 -DLIB_PICO_MULTICORE=1 -DLIB_PICO_PLATFORM=1 -DLIB_PICO_PRINTF=1 -DLIB_PICO_PRINTF_PICO=1 -DLIB_PICO_RUNTIME=1 -DLIB_PICO_STANDARD_LINK=1 -DLIB_PICO_STDIO=1 -DLIB_PICO_STDIO_UART=1 -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_UTIL=1 -DPICO -DPICO_BOARD=\"pico_w\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_CONFIG_RTOS_ADAPTER_HEADER=/home/fatdollar/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/freertos_sdk_config.h -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_USE_BLOCKED_RAM=0 -DPROJECT_NAME=bbos-dev -DPROJECT_VERSION=0.1 
```
During the build it is not until we start building  `rtos_utils.c` where the `-DUSING_PICOW` shows up:
```
[ 55%] Building C object CMakeFiles/bbos-dev.dir/rtos/rtos_utils.c.obj
/usr/bin/arm-none-eabi-gcc -DBOARD=pico_w -DCFG_TUSB_CONFIG_FILE=\"hardware_config.h\" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DCFG_TUSB_OS=OPT_OS_PICO -DCLI_USE_USB=1 -DFREE_RTOS_KERNEL_SMP=1 -DLIB_CMSIS_CORE=1 -DLIB_FREERTOS_KERNEL=1 -DLIB_PICO_BIT_OPS=1 -DLIB_PICO_BIT_OPS_PICO=1 -DLIB_PICO_DIVIDER=1 -DLIB_PICO_DIVIDER_HARDWARE=1 -DLIB_PICO_DOUBLE=1 -DLIB_PICO_DOUBLE_PICO=1 -DLIB_PICO_FIX_RP2040_USB_DEVICE_ENUMERATION=1 -DLIB_PICO_FLOAT=1 -DLIB_PICO_FLOAT_PICO=1 -DLIB_PICO_INT64_OPS=1 -DLIB_PICO_INT64_OPS_PICO=1 -DLIB_PICO_MALLOC=1 -DLIB_PICO_MEM_OPS=1 -DLIB_PICO_MEM_OPS_PICO=1 -DLIB_PICO_MULTICORE=1 -DLIB_PICO_PLATFORM=1 -DLIB_PICO_PRINTF=1 -DLIB_PICO_PRINTF_PICO=1 -DLIB_PICO_RUNTIME=1 -DLIB_PICO_STANDARD_LINK=1 -DLIB_PICO_STDIO=1 -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_UNIQUE_ID=1 -DLIB_PICO_UTIL=1 -DLIB_TINYUSB_DEVICE=1 -DPICO -DPICO_BOARD=\"pico_w\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_CONFIG_RTOS_ADAPTER_HEADER=/home/fatdollar/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/freertos_sdk_config.h -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_RP2040_USB_DEVICE_UFRAME_FIX=1 -DPICO_TARGET_NAME=\"bbos-dev\" -DPICO_USE_BLOCKED_RAM=0 -DPROJECT_NAME=bbos-dev -DPROJECT_VERSION=0.1 -DUSING_PICOW
```

I don't have time more time today but I'll continue on this later unless my findings just scream a solution to someone who sees this.  I feel like documentation is key so I just wanted to update my findings.  Pretty sure the bug is in CMAKE and not the code since manually adding `#def USING_PICOW` to the `node_dev.c` file (for testing purposes) creates the expected behavior of no `led` being listed.

mcknly, 2024-06-08:
Good find. You are correct it should be fixed in cmake, I haven't dug any deeper yet but if you identify a change to get the define passed correctly that would be great. Thanks for the documentation!

mcknly, 2024-06-17:
Fixed in #13

closed 2024-06-17 by mcknly