mcknly/breadboard-os #15: Telnet support/Wireless support.

glennswest, 2024-06-20:
Love to see telnet support and wireless support for the wireless models.
Even better option for a web server as well. 
Even better if we can script within some nodejs on client.

mcknly, 2024-06-21:
I love these ideas. We have just recently [added](https://github.com/mcknly/breadboard-os/commit/c27aa27eca6a715947a87dba96591110a02ed3d6) basic wireless support for CYW43 on PicoW, but have not done anything with it yet.

These are the steps I am envisioning:
1. Create a simple driver for connecting to a WiFi AP via CLI, and save config to the onboard flash filesystem
2. Implement a basic Telnet server (integrate an existing project would be ideal)
3. Create a BBOS service for CLI over Telnet

Once those features are in place, doing other fun things over wireless should be easier to implement.

Are you interested in contributing?

Kintar, 2024-06-24:
I'm actually working on this right now. One question I had for @mcknly is; where do you want to include the lwIP configuration? It can be bundled in the hardware/rp2040 directory, but lwIP is a fairly common library, so it might make more sense to put it at the same level as the rtos includes.

This is complicated by the fact that there are specific changes to required link libraries for pico depending on how the user wants to interact with lwIP. I could go the route of assuming we want full FreeRTOS integration by default and leave documentation breadcrumbs for people who want to switch to the non-OS or polling backends. That's probably the simplest to start with. This would involve:

- A new switch for lwIP support. Probably just `LWIP_ENABLED`
- Possibly CMake-level support for switching LWIP_SOCKETS in the lwipopts.h file? (Whether or not to include the BSD socket operations.)
- Adding an lwip directory to hold the default `lwipopts.h` file.

We could then add the pico-specific library includes to `hardware/rp2040/prebuild.cmake`, and add a new `lwip_optional.cmake` include that works at the project level to add the lwipopts header.

Thoughts?

mcknly, 2024-06-25:
@Kintar this is exciting! I'm looking forward to seeing your progress. I will convert this issue to a discussion thread, and then create a new issue+branch specific to your CYW43+lwIP implementation.