mcknly/breadboard-os #23: Error when compiling

annonymous12343, 2024-08-12:
/home/an0n1m0us/breadboard-os/services/usb_service.c:23:10: fatal error: tusb.h: No such file or directory
   23 | #include "tusb.h"
      |          ^~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/my-bbos-proj.dir/build.make:146: CMakeFiles/my-bbos-proj.dir/services/usb_service.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1882: CMakeFiles/my-bbos-proj.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

mcknly, 2024-08-13:
TinyUSB is a submodule of the Pico SDK, try: `git submodule update --init` in your pico-sdk directory

annonymous12343, 2024-08-15:
I tryied it and now it worked. Thanks!

closed 2024-08-15 by annonymous12343